fact-project/shifthelper #61: QLAs do not agree

dneise, 2016-02-03:
Please have a look at these two QLA results. 
They do not agree. 

I think it is critical for our tool, to have them agreeing. I think Max N. already knows kind of the reason for this. So maybe the question is: *Do we agree, that having the same QLA results is critical for the success of the shifthelper?*

[image]
[image]

MaxNoe, 2016-02-03:
Yes, I agree it's critical. The binslides that are there match, right? So we are discarding  bins which daniela keeps. I have to check her code again.

MaxNoe, 2016-02-03:
I removed the cut, that threw away bins that had less than 90 % fOntime of the target binning.
The result is:
[image]

MaxNoe, 2016-02-03:
I think I got it. I replaced a `>=` with a `>` and check only the last bin for the ontime.
[image]

MaxNoe, 2016-02-08:
This was close via #62 .

closed 2016-02-08 by MaxNoe