fact-project/shifthelper #65: Current Limits in config file - vs - hardocded

dneise, 2016-02-07:
Currently the Current Limits are stored in the config.ini file, however the check does not use the config file, but uses hardcoded limits.

https://github.com/fact-project/shifthelper/blob/master/fact_shift_helper/checks/webdim.py#L46

I think, the shifthelper should be a security instrument, meaning: A shifter should not be able to change accidentally these limits and then not be called. 

If a shifter gets annoyed by *false* high current calls, he has no choice but either:
 * stop observation, since there is apparently something wrong in the current measurement, or 
 * stop the shifthelper and stay awake, since he thinks observation can be continued, despite something being fishy.

So I vote for removing the current limits from the config.file and not implementing the CurrentCheck to read the limits from file.

MaxNoe, 2016-02-07:
A third possibility would be to not give the shifter access to the config file. E. G.  by using a pkg_resource and decrypting it at every start.

dneise, 2016-02-08:
I just realized this is not relevant, once the shifthelper is running on some reliable nodes instead of the shifters own box

MaxNoe, 2016-05-23:
We deicded to do it in the code

closed 2016-05-23 by MaxNoe