fact-project/shifthelper #67: no need for a long discussion - remove encrypted stuff

dneise, 2016-02-22:
I just had a discussion about the encrypted stuff we deliver with the product. There is hardly any chance, this will be accepted by the collaboration. 

As it seems gpg encrypted files can be brute forced in a month or so. I have no references for this, but I see the difference between brute forcing into a remote machine, which locks me out after a few million tries and a local file on my machine, which I can punch in for days...

When moving the shift helper to 2 or 3 machines, instead of installing it on everyones machines, there is for this fancy encryption stuff anymore anyway.

MaxNoe, 2016-02-22:
> , there is for this fancy encryption stuff anymore anyway.

no need? Or what?

dneise, 2016-02-23:
solved in 970974ed24f159b8684cee6b6fe46c327504cc5e

closed 2016-02-23 by dneise