fact-project/shifthelper #70: Find a better name

dneise, 2016-03-06:
From #69 and others I learned that `shifthelper` was not the smartest choice for a name. 

The task of the program is (greatly shortened):

**Alert the shifter, when manual interaction is definitely needed.**

But `shifthelper` implies a great deal of other tasks, which this program could do, but imho shouldn't do.
Maybe `shifthelper` is a good name for a suite of helpful programs for the shifters.

MaxNoe, 2016-04-04:
- SECFAN   Security and Flare Alert Notifier

MaxNoe, 2016-04-04:
* SELINAS **Se**curity **Li**mit **No**tification **A**lert **S**ystem

dneise, 2016-04-04:
Caller.

MaxNoe, 2016-04-04:
MoFa - Monitor Fact

MaxNoe, 2016-04-04:
PLIFAC - Python Limits and Flare Alert Caller

mackaiver, 2016-04-21:
AASSS - Automatic Alert System for Sleeping Shifters

dneise, 2016-05-26:
This was closed, since people are accustomed to the name. And I cannot stop using it.

closed 2016-05-26 by dneise