fact-project/shifthelper #72: Does not work on windows

dneise, 2016-03-08:
In case we think shifthelper should also work on windows,

maybe the 'windows' branch is a starting point.
I wonder if than branch would still run on Linux... have no machine here to test... might do it tomorrow.

https://github.com/fact-project/shifthelper/tree/windows

dneise, 2016-05-26:
The only reason for the necessity of the windows branch is our use of blessings, if I'm not mistaken.
Since in the future we do not expect users to actually look at the screen while shifthelper is running. We could drop the use of blessings entirely.

On the other hand, since in the future we plan to run it on a server as opposed to a users pc, there is no need for windows support.

MaxNoe, 2016-11-26:
Not a problem anymore with the centralised version

closed 2016-11-26 by MaxNoe