fact-project/shifthelper #77: have standalone `call` executable

dneise, 2016-05-23:
Daniela has already now a program, which she uses to detect flares. So all she needs is a program she can call, which calls her on the phone. 

There is no problem in providing a standalone with an interface like this:
```
$ call <phone number>
```

In case we can somehow make sure the program is running on gate, we could even provide something like:

```
$ call <phone number> [<text>]
```

For each call, the program could create a new twiml_file in `/home/factwww/sandbox/shifthelper` with a name reflecting:
 * time of execution
 * username
 * 8char random sequence against collisions

like `20160512_1234_dneise_ec5frwxt.xml`

containing something like:
```
<?xml version="1.0" encoding="UTF-8"?>
<Response>
    <Say voice="alice" language="en">{<text>}</Say>
</Response>
```

After a while, we can then remove these xmls based on their filename.

dadorner, 2016-05-23:
why can text only be provided when it is running on gate? 

concerning how to call the program: I would prefer to do only something like: 
`call flare-expert`
and which numbers to call in which order in that case would be taken care of by the caller program (eg retrieved from a DB) 
I think we anyhow need to deal with this type of information/procedure for the shifter, so it would not make sense to have the 'whom do I call in which order and how long do I try' procedure re-implemented in my program
also one might want to change the flare-experts and it would be good to have for this the same procedure/interface as for the shifter and not having to change the code of the flare-alert program

MaxNoe, 2016-05-23:
Twilio expects a url pointing to the xml file containing the information how the call should be handled.
The url is then requested by twilio if a call is made. 

So either the program runs on gate so it can write to a file in `factwww` or we need to have a page that accepts a `PUT` request from somewhere else.

dadorner, 2016-05-23:
I think this should not be an issue, as most machines (apart from gate) can write to /loc_data of daq (mounted as /daq on gate (only readable) and eg data)
I assume that the files are small, so we could use /daq for this

MaxNoe, 2016-05-23:
Currently the xml we use is static:

```{xml}
<?xml version="1.0" encoding="UTF-8"?>
<Response>
    <Hangup />
</Response>
```
So we end the call as soon as someone picks up to reduce costs.

MaxNoe, 2016-05-23:
@dadorner Yes, the files will be very small, few kb. 
I think dom and I meant La Palma vs. Shifter pc not necessarily gate.

dneise, 2016-05-23:
> concerning how to call the program: I would prefer to do only something like:
call flare-expert
and which numbers to call in which order in that case would be taken care of by the caller program (eg retrieved from a DB) 

Following the Linux strategy of writing programs, that do one thing, and do that one (fairly) good, we should not include this. Also it adds a dependency, which is otherwise not needed, and reduces the potential for re-usability.

If the information whom to call is in the DB, the program invoking the proposed `call` can retrieve the number from the DB and pass it to `call`.

---

However we can provide both ... first simple `call`, which just needs a number. Afterwards another program, which retrieves the number from the DB and then passes it to call in case this is to complicated to accomplish in the program you already have.

dneise, 2016-05-23:
Also I think we should drop the stuff about the text. It also adds a dependency. One needs a running web server somewhere. And in our case, the process needs to know on what machine it is running:
 * if on gate : write to /home/factwww/sandbox/shifthelper/twiml/foo/bar
 * if on another LP machine : write to /loc_data/sandbox/shifthelper/twiml/foo/bar
 * if on my laptop for testing : ???

MaxNoe, 2016-05-23:
In the long run, we could have a subdomain for the shifthelper with a flask webapp to handle this, but this will take some time I do not have right now.

dadorner, 2016-05-23:
to my understanding the shifthelper will need some procedure on which shifter to call in which order and how long to try... 
I don't see the point of re-implementing this procedure in my program for the flare-expert if the procedure is already there...

MaxNoe, 2016-05-23:
Currenty the shifthlper calls only one number, which is provided at the startup on the shifter's pc. It calls once for each alert.

MaxNoe, 2016-05-23:
See #79

MaxNoe, 2016-09-02:
This was fixed in #79

closed 2016-09-02 by MaxNoe