fact-project/shifthelper #80: shifthelper 24/7 on La Palma

dneise, 2016-05-26:
In order to run the shifthelper 24/7 on a LP machine,
the shifthelper needs to know, if there is a shift at the moment or not.
(Also who is on shift (different roles included) needs to be known.

In this issue, I want to collect some ideas, about it.

MaxNoe, 2016-05-26:
I think this would require a webservice. A little Flask app for example.

Flask I think flask is well suited because
* it can use LDAP
* it runs with our Apache (using mod_wsgi)
* it's rather light weight compared to e.g. Django

One advantage would be, that we finally have a webservice we could use for the answer_url for twilio or plivo.

MaxNoe, 2016-05-26:
Sorry, wrong button.

dneise, 2016-05-26:
Thomas Bretz proposed in the last "FACT robotics telcon" to use the smartfact error history in order to find out if there is supposed to be a shift, i.e. if `Main.js` should be running or not.

[image]

## Details about smart fact error history

Since a couple of weeks, smartfact has a new feature, which creates an alarm if no script is running, under these conditions:

  * no moon break
  * It is `dark time` or `astronomical twilight` (before midnight).

The *flag string* to be found in the error history in this case is `"No script running during datataking time."`.

So while the current condition under which the shifthelper calls a shifter is:
 * if DimCtrl not in state "Running"

It would then be connected with 
 * *and* if not *flag string* in error history.

dneise, 2016-05-26:
Regarding the "who is on shift" question the proposal was to use the entry in `calendar.Data`, or a similar table, which is also able to represent the "role" a person has.

Currently the `calendar.Data` table looks like this:

    mysql> explain calendar.Data;
    +-------+--------------+------+-----+---------+-------+
    | Field | Type         | Null | Key | Default | Extra |
    +-------+--------------+------+-----+---------+-------+
    | y     | mediumint(9) | YES  |     | NULL    |       |
    | m     | tinyint(4)   | YES  |     | NULL    |       |
    | d     | tinyint(4)   | YES  |     | NULL    |       |
    | u     | varchar(255) | YES  |     | NULL    |       |
    | x     | tinyint(1)   | NO   |     | 0       |       |
    +-------+--------------+------+-----+---------+-------+

The `(y,m,d)` columns encode a date. the `u` column stores a user name or institute name or the string "moon". `x` encodes a boolean, specifying whether a shift was an expert shift or a normal shift.

    +------+------+------+----------+---+
    | y    | m    | d    | u        | x |
    +------+------+------+----------+---+
    | 2016 |    4 |   11 | walter   | 0 |
    | 2016 |    4 |   11 | balbo    | 0 |
    | 2016 |    4 |   12 | walter   | 0 |
    | 2016 |    4 |   12 | balbo    | 0 |
    | 2016 |    4 |   13 | balbo    | 0 |
    | 2016 |    4 |   14 | balbo    | 0 |
    | 2016 |    4 |   15 | balbo    | 0 |
    | 2016 |    4 |   16 | balbo    | 0 |
    | 2016 |    4 |   17 | dorner   | 1 |
    | 2016 |    4 |   17 | moon     | 0 |
    | 2016 |    4 |   18 | moon     | 0 |
    | 2016 |    4 |   19 | moon     | 0 |
    | 2016 |    4 |   20 | moon     | 0 |
    | 2016 |    4 |   21 | moon     | 0 |
    | 2016 |    4 |   22 | moon     | 0 |
    | 2016 |    4 |   23 | moon     | 0 |
    | 2016 |    4 |   24 | dbaack   | 0 |
    | 2016 |    4 |   25 | dbaack   | 0 |
    | 2016 |    4 |   26 | dbaack   | 0 |
    | 2016 |    4 |   27 | dbaack   | 0 |
    | 2016 |    4 |   28 | smueller | 0 |
    | 2016 |    4 |   28 | ETHZ     | 0 |
    | 2016 |    4 |   29 | smueller | 0 |
    | 2016 |    4 |   29 | ETHZ     | 0 |
    | 2016 |    4 |   30 | ETHZ     | 0 |
    | 2016 |    4 |   30 | smueller | 0 |
    | 2016 |    4 |   31 | ETHZ     | 0 |
    | 2016 |    4 |   31 | smueller | 0 |
    +------+------+------+----------+---+

MaxNoe, 2016-05-26:
There is a mysql data type called `date`!

dneise, 2016-05-26:
I am fully aware of that. I didn't design this stuff. Just trying to explain it.

dneise, 2016-05-26:
In order for the shifthelper to find out from this, what number to call, one could make use of the `logbook.users` and `logbook.userfields` tables.

`logbook.users` is pretty long, these columns are maybe interesting:

    mysql> describe users;
    +------------------------+----------------------+------+-----+---------+----------------+
    | Field                  | Type                 | Null | Key | Default | Extra          |
    +------------------------+----------------------+------+-----+---------+----------------+
    | uid                    | int(10) unsigned     | NO   | PRI | NULL    | auto_increment |
    | username               | varchar(120)         | NO   | UNI |         |                |
    | email                  | varchar(220)         | NO   |     |         |                |
    | website                | varchar(200)         | NO   |     |         |                |
    | icq                    | varchar(10)          | NO   |     |         |                |
    | aim                    | varchar(50)          | NO   |     |         |                |
    | yahoo                  | varchar(50)          | NO   |     |         |                |
    | msn                    | varchar(75)          | NO   |     |         |                |
    | away                   | int(1)               | NO   |     | 0       |                |
    | awaydate               | int(10) unsigned     | NO   |     | 0       |                |
    | returndate             | varchar(15)          | NO   |     |         |                |
    | awayreason             | varchar(200)         | NO   |     |         |                |
    +------------------------+----------------------+------+-----+---------+----------------+

    mysql> describe userfields;
    +-------+------------------+------+-----+---------+-------+
    | Field | Type             | Null | Key | Default | Extra |
    +-------+------------------+------+-----+---------+-------+
    | ufid  | int(10) unsigned | NO   | PRI | 0       |       |
    | fid1  | text             | NO   |     | NULL    |       |
    | fid3  | text             | NO   |     | NULL    |       |
    | fid4  | text             | YES  |     | NULL    |       |
    | fid5  | text             | YES  |     | NULL    |       |
    | fid6  | text             | YES  |     | NULL    |       |
    | fid7  | text             | YES  |     | NULL    |       |
    | fid8  | text             | YES  |     | NULL    |       |
    +-------+------------------+------+-----+---------+-------+

these `fid#` column names are not really telling, so here I performed this query:
`select users.username, userfields.* from users join userfields on users.uid=userfields.ufid where users.username ="dneise";`

And learned this:
 * fid1 : institute name
 * fid3 : gender
 * fid4 : office number
 * fid5 : mobile number
 * fid6 : empty - maybe "Comment"
 * fid7 : skype name
 * fid8 : Contact Info  <-- only this field is used by [shiftinfo](http://fact-project.org/dch/shiftinfo.php)

MaxNoe, 2016-05-26:
We need an icq integration!!!!!

dneise, 2016-05-26:
I see at the moment at least one problem with Thomas proposal, using the error history of smartfact, to find out, whether there is a shift or not at the moment.

The error history can contain errors several days old. This leads to the following problem for the shifthelper.

## Example:

 * During the night Main.js threw an exception and stopped.
 * smartfact alarmed, then shifthelper called.
 * The shifter fixed it. No other alarms occurred for  the rest of the night.
 * In the morning, the shutdown went smooth. Last step: stop Main.js.

Immediately and constantly the shifthelper calls the shifter. Why?
Because the error history contains the "No script running" alarm message.

dneise, 2016-05-26:
Other possible solutions for this "calling in case Main.js no running" matter are:

 * shifthelper checks in `calendar.Data` if moon and what time it is, done.
 * smartfact dispatches the phone calls to the shifter(s).

MaxNoe, 2016-05-26:
According to wikipedia, the different twilights are defined as:
[image]

Using `astropy` we could implement the twilight check ourselves like this:

```
from astropy.coordinates import EarthLocation, get_sun, AltAz
from datetime import date
import astropy.units as u
from astropy.time import Time
import numpy as np

orm = EarthLocation.of_site('Roque de los Muchachos')

def twilight(night):
    t0 = Time('{}-{}-{} 15:00'.format(night.year, night.month, night.day))
    times = t0 + np.arange(0, 20, 1/60) * u.hour

    sun = get_sun(times)
    frame = AltAz(obstime=times, location=orm)
    sun_altaz = sun.transform_to(frame)

    astro_twilight = times[sun_altaz.alt.deg < -12]

    return astro_twilight.min(), astro_twilight.max()

```

dneise, 2016-05-26:
yes we could calculate it as you've shown, or read it from the smartfact website as e.g. in

```{python}
    import smart_fact_crawler
    smart_fact_crawler.sun()
    # result -->
    {'End_of_astro_twilight': datetime.datetime(2016, 5, 27, 5, 15, 0, 872000),
     'End_of_dark_time': datetime.datetime(2016, 5, 27, 4, 40, 0, 872000),
     'End_of_day_time': datetime.datetime(2016, 5, 26, 20, 1, 0, 872000),
     'End_of_nautic_twilight': datetime.datetime(2016, 5, 27, 5, 47, 0, 872000),
     'Start_of_astro_twilight': datetime.datetime(2016, 5, 26, 20, 59, 0, 872000),
     'Start_of_dark_time': datetime.datetime(2016, 5, 26, 21, 34, 0, 872000),
     'Start_of_day_time': datetime.datetime(2016, 5, 27, 6, 14, 0, 872000),
     'Start_of_nautic_twilight': datetime.datetime(2016, 5, 26, 20, 27, 0, 872000),
     'Time_Stamp': datetime.datetime(2016, 5, 26, 13, 1, 31, 872000)}
```

Both solutions have their advantages, None of them introduces new dependencies, which makes it difficult for me to chose one over the other.

MaxNoe, 2016-05-26:
Ah, forgot about that. That would be better as it does not introduce a new dependency (currently we do not use astropy)

dneise, 2016-05-26:
At the moment I am working on a solution according to my own proposal number two, for a test run on LaPalma. (just saying, so we are not working on stuff simultaneously)

MaxNoe, 2016-05-26:
No branch pushed? Why the secretness?

dneise, 2016-05-26:
no code existing yet

MaxNoe, 2016-09-02:
We decided on the GM that shifthelper will be running outside La Palma to be able to check if the Island is still alive.

closed 2016-09-02 by MaxNoe