fact-project/shifthelper #97: local replica of DB

dneise, 2016-09-14:
shifthelper needs always access to DB, so we create a local replica of selected parts (if not all) of our mysql DB on lapalma.

dneise, 2016-09-14:
we thought about replicating into a local sqlite3 DB, but this create synchronization issues, so we go for a normal mysql DB.

dneise, 2016-09-14:
we could use mysql intrinsic replication or roll out our own:
 - every 30 minutes ... read entire remote DB and push it into our local replica.

not sure what is better

dneise, 2016-09-29:
Working on this .... have a look at #113

MaxNoe, 2016-09-29:
As it propably runs in the same Computer AS the Webinterface we could use the same mysql db

dneise, 2016-09-29:
It is using the same db, but you can't know this because stupid me forgot to push the new config file. Sorry

dneise, 2016-10-23:
working now

closed 2016-10-23 by dneise