fact-project/shifthelper #110: Users who have 2 phone_numbers

dneise, 2016-09-15:
There are people who need to have two phone numbers simultaneously, since they continuously travel between places where either of the two numbers have connection.

Since this happens often and maybe sometimes on short notice, they cannot be asked to continuously changes these numbers in their contact info (logbook).

Two things are needed

 * the people need to be able to provide two (many) numbers
 * the notifiers need to be able to simply call them on all their numbers simultaneously.

dneise, 2016-11-03:
Okay so I send Daniela a mail, so this issue can move a bit forward

dneise, 2017-03-06:
Users can now change their number in the logbook and the shifthelper will call this modified number within the next 5 minutes. 

This was solved in acad86656185e8fbc64c4dd9a2b4dad2637a75d1

closed 2017-03-06 by dneise

MaxNoe, 2017-03-06:
There is also an abstract about using call redirection in http://github.com/fact-project/nightshift_guide