fact-project/shifthelper #118: no call - during test 21.10.16

dneise, 2016-10-22:
Tonight around 04:35 the old shifthelper correctly called the shifter, but the new one did not.

old shifthelper log:
```
04:34:30 - WARNING - shift_helper | humidity_outside >= 98 %: 98.0 %
04:35:32 - WARNING - shift_helper | humidity_outside >= 98 %: 98.0 %
04:36:33 - WARNING - shift_helper | humidity_outside >= 98 %: 98.1 %
04:37:29 - WARNING - shift_helper | humidity_outside >= 98 %: 98.1 %
```

For the complete new shifthelper log, see logfile on ihp-pc41.
As one can e.g. see here
```
2016-10-22 04:37:11,536 - shifthelper.checks - DEBUG - HumidityCheck: humidity:98.1 lid_status:Open
```
the value > 98 was correctly seen also by the new shifthelper, but no further action was done.

dneise, 2016-10-22:
In this context, I had an idea. I would like to be able to *inject* fake-situations into the input data stream of the new shifthelper, which should lead to a call, or at least to an `info`-log entry. 

For this, I would propose to put a button on the shifthelper_webinterface, named `inject warning` and `inject info`, which will do that. The new shifthelper will get an additional Checker, which simply checks something like `https://shifthelper_webinterface/injects` for new injections. As soon as a new injection pops up there, this `FactIntervalChecker` will simply execute its own `self.warn` or `self.info` methods and thus serve as a nice test. 

There is possibility for abuse here, since this will result in a call to the current shifter, anybody who presses these buttons annoys the current shifter. This is a task to be solved...

dneise, 2016-10-22:
[log_on_stdout.txt](https://github.com/fact-project/shifthelper/files/545788/log_on_stdout.txt)

I must have made an error in the logging settings. I just saw some exceptions on shifthelpers stdout, which is normally empty. They explain, that no call went out, since `/alerts` was not available, and the shifthelper was not able to find out, if the alerts were acknowledged or not.

In this case, the shifthelper should call anyway, but it threw an exception and did not call.
Not good. We need a test to ensure this is not happening anymore and we need to fix this.

MaxNoe, 2016-10-22:
See #119

dneise, 2016-10-23:
Unfortunately #119 did not fix this :-(

old shifthelper logfile here: [shifthelper_2016-10-22_log.txt](https://github.com/fact-project/shifthelper/files/546233/shifthelper_2016-10-22_log.txt)

Shows high humidity again around 4:35h

---
new shifthelper log shows again, humidity was seen high, but no warning or info is generated:
```
2016-10-23 04:38:21,944 - shifthelper.checks - DEBUG - RelativeCameraTemperatureCheck: relative_temperature:5.74
2016-10-23 04:38:37,523 - shifthelper.checks - DEBUG - HumidityCheck: humidity:98.5 lid_status:Open
2016-10-23 04:39:01,413 - shifthelper.checks - DEBUG - BiasChannelsInOverCurrent: bias_state:VoltageOn
2016-10-23 04:39:01,618 - shifthelper.checks - DEBUG - ContainerTooWarm: container_temperature:26.3
```

-----

Again there are exceptions printed to stdout an not logged. What did I do wrong here?
[shifthelper_stdout_2016_10_21.txt](https://github.com/fact-project/shifthelper/files/546236/shifthelper_stdout_2016_10_21.txt)

But all it contains is this a gazillion times:
```
Exception while running check
Traceback (most recent call last):
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/requests/packages/urllib3/connection.py", line 142, in _new_conn
    (self.host, self.port), self.timeout, **extra_kw)
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/requests/packages/urllib3/util/connection.py", line 98, in create_connection
    raise err
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/requests/packages/urllib3/util/connection.py", line 88, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/requests/packages/urllib3/connectionpool.py", line 595, in urlopen
    chunked=chunked)
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/requests/packages/urllib3/connectionpool.py", line 363, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/http/client.py", line 1106, in request
    self._send_request(method, url, body, headers)
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/http/client.py", line 1151, in _send_request
    self.endheaders(body)
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/http/client.py", line 1102, in endheaders
    self._send_output(message_body)
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/http/client.py", line 934, in _send_output
    self.send(msg)
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/http/client.py", line 877, in send
    self.connect()
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/requests/packages/urllib3/connection.py", line 167, in connect
    conn = self._new_conn()
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/requests/packages/urllib3/connection.py", line 151, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
requests.packages.urllib3.exceptions.NewConnectionError: <requests.packages.urllib3.connection.HTTPConnection object at 0x7fd2a801a9e8>: Failed to establish a new connection: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/requests/packages/urllib3/connectionpool.py", line 640, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/requests/packages/urllib3/util/retry.py", line 287, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
requests.packages.urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='localhost', port=5000): Max retries exceeded with url: /alerts (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnection object at 0x7fd
2a801a9e8>: Failed to establish a new connection: [Errno 111] Connection refused',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
[... 4 lines cut ...]
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/retrying.py", line 49, in wrapped_f
    return Retrying(*dargs, **dkw).call(f, *args, **kw)
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/retrying.py", line 212, in call
    raise attempt.get()
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/retrying.py", line 247, in get
    six.reraise(self.value[0], self.value[1], self.value[2])
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/six.py", line 686, in reraise
    raise value
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/retrying.py", line 200, in call
    attempt = Attempt(fn(*args, **kwargs), attempt_number, False)
  File "/fast_loc/home/factshifthelper/shifthelper/shifthelper/checks.py", line 44, in all_recent_alerts_acknowledged
    all_alerts = requests.get('http://localhost:5000/alerts').json()
[... cut ...]
    r = adapter.send(request, **kwargs)
  File "/fast_loc/home/factshifthelper/anaconda3/lib/python3.5/site-packages/requests/adapters.py", line 487, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='localhost', port=5000): Max retries exceeded with url: /alerts (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnection object at 0x7fd2a801a9e8>: Fai
led to establish a new connection: [Errno 111] Connection refused',))
```

MaxNoe, 2016-10-23:
Ah, dam. requests uses it's own exception:

```
In [2]: import requests.exceptions
In [3]: issubclass(requests.exceptions.ConnectionError, ConnectionError)
Out[3]: False
```

I think the log on standard out is from the `requests` logger instance. We might need to configure it explicitly.

dneise, 2016-10-23:
solved by #120

closed 2016-10-23 by dneise