fact-project/shifthelper #123: Multiple Alerts on Website due to same reason

dneise, 2016-10-23:
I just got a couple of cals, since Main.js was not running. 
I think it is enough to show the Alert only once on the website ... 
[image]

dneise, 2016-11-03:
do you disagree. Any opinion on this?

MaxNoe, 2016-11-03:
I think I could do a groupby in the DB to show anf acknowledge all alerts of the same kind in one click.

dneise, 2016-11-03:
Is showing only the last alert per check maybe enough?

MaxNoe, 2016-11-03:
This semantically different, maybe yes.

dneise, 2016-11-03:
Have a look at the page, whats the point in even showing all of them? Maybe one different alert hides inside many equal ones ... who knows ... 

[image]

dneise, 2017-03-06:
@MaxNoe can you remind me of the status of this? Can we maybe close this?

MaxNoe, 2017-03-06:
As this is just cosmetics, I did not yet make any effort to tackle this.

I would leave it open, but this is just nice to have.