fact-project/shifthelper #128: Supervisor

dneise, 2016-10-31:
We agreed, the new shifthelper should be running under supervision, so it is restarted, when it dies.
At the moment we are not working on this.

Is anybody willing to work on this part and start the currently running test version on `ihp-pc41.ethz.ch` inside a supervisor of his choice?

MaxNoe, 2016-10-31:
A possible solution would be a docker container with `--restart=always`, this would also have the advantage, that we could `link` the mysql database to the container and thus do not need to expose the ports on the host machine.

dneise, 2016-10-31:
Any solution is fine with me. Would start setting it up and somehow leave documentation for us non-docker-savy people on how to start/stop/restart it, in case we need to?

MaxNoe, 2016-12-03:
At the moment this is down by the "restart: always" option of docker.

closed 2016-12-03 by MaxNoe