fact-project/shifthelper #136: Website did not show alerts for shifter

dneise, 2016-11-05:
This morning the shifthelper called a testshifter who had the master branch running simulatneosly. 
So incoming calls cannot be distinguished. He looked at the shifthelper webinterface and no alerts were shown. Even after refreshing the page, no alerts were shown.

I wasn't able to reproduce this later. Not sure how to hunt this bug down.
We have to see tomorrow morning how it behaves.

MaxNoe, 2016-11-05:
Which alerts where not shown? The website currently shows 4 shifter not awake alerts. These are also the only ones in the database for today.

dneise, 2016-11-07:
Exactly those were not shown, if I remember correctly. As I said: I was not able to reproduce this, i.e. for me everything was shown right away correctly.

MaxNoe, 2016-12-03:
I will close this for now because it did not show up again and might just have been a call of the old shifthelper.

closed 2016-12-03 by MaxNoe