fact-project/shifthelper #144: developer is called, when shifter does not put telegram_id into contacts

dneise, 2016-11-10:
Is putting ones telegram_id into the contact info database optional? 
At the moment the developer is called, since this creates an exception. 

In case putting the telegram_id is not optional, since we regard this as a 2nd independent communication channel, alerting the developer is fine, so the developer can ask to shifter to put her telegram_id. 

Else we should catch this particular case and not raise it the exception up to the developer. 
Where should one catch it? In custos? No, it's a FACT decision not a custos decision, right?

dneise, 2016-11-10:
Should we do it like this:
```{python}
class FactTelegramNotifier(custos.TelegramNotifier):
    def notify(self, recipient,  msg):
        if not recipient is None:
            super().notify(recipient,  msg)
```

MaxNoe, 2016-11-10:
It's not encessary to create a new NOtifier for this, just return an empty list instead of None

MaxNoe, 2016-11-10:
But custos could also do nothing `if recipient is None`, a recipient which is `None` doesn't make to much sense.

dneise, 2016-11-10:
Ah you mean in `__main__.telegram_book`, right? so instead of

    def telegram_book(category):
        return [whoisonshift().telegram_id]

we could do:

    def telegram_book(category):
        telegram_id = whoisonshift().telegram_id
        if telegram_id is None:
            return []
        else:
            return [telegram_id]

MaxNoe, 2016-11-16:
This got fixed in #145

commit b2733e80 (2016-11-16): Return empty list instead of None, fix #144
closed 2016-11-16 by MaxNoe