fact-project/shifthelper #147: Shutdown checklist check fails when there is no shift

MaxNoe, 2016-11-15:
The check expects that the checklist is filled, but obviously isn't:

```
shifthelper_1               | 2016-11-15 13:23:00,404 - WARNING - custos.notify.log - notify - Message: no shift at the moment and 
shifthelper_1               | Last Shutdown is already 10min past and 
shifthelper_1               | checklist not filled    For: all
```

dneise, 2016-11-16:
I think this is no bug.

---

These are the important involved conditions:
```{python}
def is_last_shutdown_already_10min_past():
    '''Last Shutdown is already 10min past'''
    return get_last_shutdown() + timedelta(minutes=10) < datetime.utcnow()
```
This one is always true, when any last shutdown is more than 10 minutes past.

```{python}
def is_checklist_not_filled():
    '''checklist not filled'''
return get_last_parking_checklist_entry() < get_last_shutdown()
```

This is true when there is no checklist entry for the last shutdown. And there was not checklist entry for the last shutdown. Because stupid me didn't fill the shutdown checklist, after the shudown after my tests during moon time.

MaxNoe, 2016-11-16:
Ah, ok. Good.

closed 2016-11-16 by MaxNoe