fact-project/shifthelper #154: 3 times: DriveInErrorDuringDataRun - ERROR

dneise, 2016-11-27:
* 2016-11-26 19:09:30
 * 2016-11-26 19:19:31
 * 2016-11-26 21:35:20

traceback:
```
Exception while running check. Traceback: Traceback (most recent call last): File "/opt/conda/lib/python3.5/site-packages/custos/checks/__init__.py", line 83, in wrapped_check self.check(*args, **kwargs) File "/opt/conda/lib/python3.5/site-packages/shifthelper/checks.py", line 41, in check if all([f() for f in self.checklist]): File "/opt/conda/lib/python3.5/site-packages/shifthelper/checks.py", line 41, in <listcomp> if all([f() for f in self.checklist]): File "/opt/conda/lib/python3.5/site-packages/wrapt/wrappers.py", line 522, in __call__ args, kwargs) File "/opt/conda/lib/python3.5/site-packages/shifthelper/debug_log_wrapper.py", line 9, in log_call_and_result result = wrapped(*args, **kwargs) File "/opt/conda/lib/python3.5/site-packages/shifthelper/conditions.py", line 95, in is_data_run sfc.main_page().system_status AttributeError: 'NoneType' object has no attribute 'groups' 
```

MaxNoe, 2016-11-27:
This is actually the same as #157 and i opened the issue here: https://github.com/fact-project/smart_fact_crawler/issues/18

dneise, 2016-11-27:
closed as a dupe of #157

closed 2016-11-27 by dneise