fact-project/shifthelper #158: too long messages for twilio to read

dneise, 2016-11-27:
twilio was not able to call me since the error message was too long.
we should maybe simply cut the message down to 4000 chars.

```
2016-11-26 23:59:59,425 - shifthelper.notifiers - ERROR - Could not notifiy recipient +41774528842
Traceback (most recent call last):
  File "/opt/conda/lib/python3.5/site-packages/shifthelper/notifiers.py", line 88, in handle_message
    self.notify(phone_number, msg)
  File "/opt/conda/lib/python3.5/site-packages/shifthelper/notifiers.py", line 27, in notify
    super().notify(recipient, msg)
  File "/opt/conda/lib/python3.5/site-packages/custos/notify/twilio.py", line 71, in notify
    self.place_call(recipient, url)
  File "/opt/conda/lib/python3.5/site-packages/custos/notify/twilio.py", line 56, in place_call
    timeout=self.ring_time,
  File "/opt/conda/lib/python3.5/site-packages/twilio/rest/resources/calls.py", line 154, in create
    return self.create_instance(kwargs)
  File "/opt/conda/lib/python3.5/site-packages/twilio/rest/resources/base.py", line 365, in create_instance
    data=transform_params(body))
  File "/opt/conda/lib/python3.5/site-packages/twilio/rest/resources/base.py", line 200, in request
    resp = make_twilio_request(method, uri, auth=self.auth, **kwargs)
  File "/opt/conda/lib/python3.5/site-packages/twilio/rest/resources/base.py", line 164, in make_twilio_request
    uri=resp.url, msg=message, code=code)
twilio.rest.exceptions.TwilioRestException: HTTP 400 error: Url must have fewer than 4000 characters
```

MaxNoe, 2016-11-27:
I think we should just change to the hangup xml, as the shifters should not take the calls in the near future.

dneise, 2016-11-27:
Ah I just see this is both an issue for custos. I move the issue there

dneise, 2016-11-27:
Ah no, I see we can set `self.twiml == 'hangup'` in the `FactTwilioNotifier` and thus steer the behaviour of the mother class .. doing that also now.

PR #161 (2016-11-27): use the hangup xml
trying to fix #158
closed 2016-12-03 by MaxNoe