fact-project/shifthelper #162: Let's use the startup_checklist to generate a call

dneise, 2016-11-27:
The last shifter did not startup main.js ~20minutes before the scheduled startup, but about 3 minutes later, because there was something urgent to do. 

I think we should generate a call, when the startup checklist is not filled until 20minutes before the scheduled startup, this might have reminded him early enough ...

jebuss, 2019-01-07:
Last night I forgot that I was on shift. The shift helper called me but at the time he called me the scheduled startup time was already passed. So I up this issue

MaxNoe, 2019-01-07:
If that happend, it's not correct. The shifthelper should have called you ten minutes before the scheduled startup.

MaxNoe, 2019-01-07:
We need to investigate what happend here, because the startup was scheduled for 19:08, but the shifthelper only called at 19:10. It should have called as early as 18:58.

MaxNoe, 2019-01-07:
Checking the code again, this is not the case.
Should we just add a `- timedelta(minutes=15)` to `is_shift_at_the_moment`? Then it would call fifteen minutes before startup if main.js is not running.