fact-project/shifthelper #166: shifthelper does not call shifter - directly fallback shifter. tescall insufficient?

dneise, 2016-12-05:
This morning (05.12.16) the fallback shifter got three calls and the shifter none. This is one problem.

The testcall correctly results in a call to the shifter. This is a more fundamental problem, in my opinion. The testcall at the moment only checks if the current website user typed in her phone number correctly into the database. That is, it neither tests if the current website user is on shift today (has led to problems in the past) nor it tests if currently the system is working in a way that Warning messages lead to calls to the shifter. 

I understand, these are 3 different problems:

 * number correctly stored in DB.
 * Does the system agree with me, that I am today on shift?
 * Does the system really really dispatch calls to me or maybe to the fallback shifter instead of me?

So I hear already people saying: there should be 3 different tests for this. 
However one single functional test making sure that for tonight the system might work for me, might not be entirely useless. I propose to push a dummy warning message into the queue, when somebody wishes to do so. 

There could be a button on the website saying: "create test Alert" or so. Which after pressing it checks if the user who pressed the button is identical with the current shifter if yes, we go on, if not we are a little too verbose and actually tell the user who we think is on shift tonight, and that we thing the current website user is not tonights shifter. 

So if the user was the shifter: we return this little json on GET requests on the "/testAlert" endpoint:
```
{ user: "current user name",
  time: "timestamp when the create-test-alert button was pressed",
  uuid: "maybe? to avoid multiple alerts from the same test-alert?"
}
```

A new check needs to be implemented in the shifthelper which checks this entpoint every 5minutes, and in case there is a new test-alert-json thing, it creates a warning-message. Which in turn results in a call to ... well .. either the shifter ... or the fallback shifter ...

MaxNoe, 2016-12-05:
Please have a look at #163, this addresses some of the problems mentioned here.
Especially it fixes the behaviour, that if the shifter misses one call, he gets never called again but the calls got directly to the backup shifter instead.

MaxNoe, 2016-12-14:
Closed since #163 is merged

dneise, 2016-12-16:
This is again one of my famous "double" issues ... I'll open a new one only regarding this test call issue.

dneise, 2016-12-16:
Okay so I made a new issue #176 and close this again.

closed 2016-12-16 by dneise