fact-project/shifthelper #187: Retrying not working as intended for smart_fact_crawler

MaxNoe, 2016-12-20:
The retrying as defined in `shifthelper.retry_smart_fact_crawler` does not work as intended.

I suspect this is a bug in retrying, as the decorator is not behaving as it should (applying it as function instead of as decorator).

The result is a "Never giveup, never surrender, no back off" mode for the retrying.

dneise, 2016-12-20:
You are right 

     retry(sfc.status, **kwargs)

Has to be

    retry(**kwargs)(sfc.status)

Sorry that was my fault, I guess

dneise, 2016-12-21:
I don't see that this is the fault of retrying but this is a misunderstanding of the python syntax for decorators with arguments, d.f.
https://docs.python.org/3/reference/compound_stmts.html#function-definitions

So I don't think we *have to* use the syntax you proposed. One *can* still use the shorter way.... just saying. 

The point is. Its great you found that out. I also ran into this in the past and thought I had fixed it. 
What I did not do is: 
 - write a test that shows the problem
 - fix the problem
 - enjoy that the test is passing.
 - go party.

**all thumbs up**

commit 0e7d1c9d (2017-01-13): Fix retry behaviour, fix #187
closed 2017-01-13 by dneise