fact-project/shifthelper #192: strange error

dadorner, 2016-12-24:
see here: https://www.fact-project.org/logbook/showthread.php?tid=4787&pid=23612#pid23612

furthermore two test calls and one dummy alert didn't work today

dneise, 2017-01-06:
Thanks for reporting this here. Chances are high, I would have overlooked it in the logbook.

The Error you saw there on the shifthelper web interface was a "developer-level" error. So Max was called not you, being the shifter.

In this case, the error was that a mysql DB was not accessible for the "ParkingChecklistCheck" at 20:00UTC. Like every check, also this check is executed every 5 minutes ... in this case just to find out that there was no shutdown recently, so it goes back to sleep.
In order to find out if it is time to check for the parking checklist, it needs to know the schedule .. which it takes from a DB table. This table was not accessible, so it cried for help.

After 5 minutes the check is executed again and that time it worked fine, so Max must have decided to not take any action besides making a mental note to find out why sometimes a DB is not accessible.

So from the security point of view:
 - somebody was called 
 - after 5 minutes the problem did not occur again, so we were fine.

dneise, 2017-01-06:
> furthermore two test calls and one dummy alert didn't work today

What Thomas told all of us multiple times in 2011 for the FACT++ programs holds for basically all kinds of software. When reporting a problem, include the time, so developers have a chance to find the according section in the log files. 

According to the Twilio Logs you received calls at:

2016-12-25 06:38:15 UTC
2016-12-25 00:45:38 UTC
2016-12-24 20:15:14 UTC
2016-12-24 20:12:56 UTC
2016-12-24 17:02:40 UTC
2016-12-24 16:05:14 UTC

dneise, 2017-01-06:
Here is the first call due to a dummy allert:
 
```
2016-12-24 17:02:29,638 - shifthelper.conditions - DEBUG - is_dummy_alert_by_shifter -> True
2016-12-24 17:02:29,656 - shifthelper.checks - DEBUG - Giving message status WARNING
2016-12-24 17:02:29,671 - shifthelper.notifiers - DEBUG - Got a message
2016-12-24 17:02:29,671 - shifthelper.notifiers - DEBUG - Message is over alert level
2016-12-24 17:02:29,671 - shifthelper.notifiers - DEBUG - Getting phone number of primary shifter
2016-12-24 17:02:29,682 - shifthelper.tools.shift - DEBUG - Found shifter dorner
2016-12-24 17:02:29,682 - shifthelper.notifiers - INFO - Calling <number removed>
2016-12-24 17:02:29,682 - custos.notify.twilio - DEBUG - Received message
2016-12-24 17:02:29,683 - custos.notify.twilio - DEBUG - Placing call
```

dneise, 2017-01-06:
after that first call you must have pressed the acknowledge button, so the shifthelper did not call anymore.

dneise, 2017-01-06:
16:05h must have been a test call.

dneise, 2017-01-06:
20:12h must have been this testcall: https://www.fact-project.org/logbook/showthread.php?tid=4787&pid=23613#pid23613

20:15h was main.js not running:
2016-12-24 20:14:57,467 - custos.notify.log - WARNING - Check: MainJsStatusCheck - Message: There is a shift at the moment and 
Main.js is not running

So around 0:45 you must have made another test call?

2016-12-25 06:38:01,660 - custos.notify.log - WARNING - Check: IsUserAwakeBeforeShutdown - Message: There is a shift at the moment and 
20min before shutdown and 
Parker not Awake

dadorner, 2017-01-06:
> So from the security point of view:
> 
>   somebody was called
>   after 5 minutes the problem did not occur again, so we were fine.

this is good to know. 
It would be even better, if this was communicated also to the shifter, see https://github.com/fact-project/shifthelper_webinterface/issues/19

dadorner, 2017-01-06:
> 2016-12-24 16:05:14 UTC
this was the second test call - at 15:55 one didn't work

> 2016-12-24 17:02:40 UTC
this was the second dummy alert - the first one did not work 

> 2016-12-24 20:12:56 UTC
this was the fourth test call - the third one (shortly before that) did not work

> 2016-12-24 20:15:14 UTC
call as main.js had stopped (see logbook) 

> 2016-12-25 00:45:38 UTC
this was the fourth dummy alert - the third (shortly before that) did not work 

> 2016-12-25 06:38:15 UTC
this was the call 20 min before shutdow (I was not aware of the functionality of the "I'm awake button")

dneise, 2017-01-09:
We also saw now the "Internal Server Error" on the shifthelper webinterface once. 
It was caused by a broken pipe to the mysql server.

dneise, 2017-01-09:
I opened an issue in the webinterface repo and I think this can be closed now?

MaxNoe, 2017-01-09:
Yep, this has nothing to do with the actual shifthelper code

closed 2017-01-09 by MaxNoe