fact-project/shifthelper #198: One Flare - One Call

madoee, 2017-01-09:
Currently, Shifthelper repeatedly calls once the QLA has yielded Flare Alert limits, even if it is the same flare. This results in several calls for the same flare.

MaxNoe, 2017-01-09:
It shouldn't do that allready. It should only call again for the same source if the rate is higher than before. Was that the case?

dneise, 2017-01-09:
Thanks for reporting @madoee.
@MaxNoe  have a look here https://ihp-pc41.ethz.ch/ then you see what is the case and what not.
[image]

MaxNoe, 2017-01-09:
Mmh, that's strange. I will investigate

MaxNoe, 2017-02-06:
Fixed by #211

closed 2017-02-06 by MaxNoe