fact-project/shifthelper #221: developer call, due to missing table; just once.

dneise, 2017-03-06:
got this exception only once .. worked afterwards ... not sure what might be the reason.
```
Exception while running check ParkingChecklistFilled: (pymysql.err.ProgrammingError) (1146, "Table 'shifthelper.factdata_Schedule' doesn't exist") [SQL: '\n        SELECT * FROM factdata_Schedule AS S\n        WHERE\n            S.fMeasurementTypeKey = 6\n        AND\n            S.fStart < "2017-03-06 10:43:54"\n        ORDER BY S.fStart DESC\n        LIMIT 1\n        ']
```

MaxNoe, 2017-03-06:
Got this also last night, might be that the request was done right during the cloning?

dneise, 2017-03-06:
Might be ... I've no idea how long the time-out or retry-time for this check is, we should increase it maybe  ...

MaxNoe, 2017-03-06:
There is no retry at all for the request to the cloned db

dneise, 2017-03-06:
okay, maybe we should set one up then ...

dneise, 2017-03-23:
Isn't this fixed since #225; #226 ?

MaxNoe, 2017-03-23:
Yes.

closed 2017-03-23 by MaxNoe