fact-project/shifthelper #228: Test call does not work on first try

relleums, 2017-03-25:
After logging in to the shifthelper web interface on ihp-pc41, I request a test call.
Both yesterday and today this test call was not established in the first try. I received an internal server error page. (sorry, I do not have a screen shot) The error page says, that the server might be overloaded. On the second try the test call works. 

My browser: Firefox 47.0

MaxNoe, 2017-03-26:
Issue for the webinterface, no the helper

MaxNoe, 2017-03-26:
But it seems to be a mysql hickup:
```
shifthelper_webinterface_1  | [2017-03-26 12:56:43,256] ERROR in app: Exception on /testCall [GET]
shifthelper_webinterface_1  | Traceback (most recent call last):
shifthelper_webinterface_1  |   File "/opt/conda/lib/python3.5/site-packages/pymysql/connections.py", line 1004, in _write_bytes
shifthelper_webinterface_1  |     self._sock.sendall(data)
shifthelper_webinterface_1  |   File "/opt/conda/lib/python3.5/site-packages/eventlet/greenio/base.py", line 390, in sendall
shifthelper_webinterface_1  |     tail = self.send(data, flags)
shifthelper_webinterface_1  |   File "/opt/conda/lib/python3.5/site-packages/eventlet/greenio/base.py", line 384, in send
shifthelper_webinterface_1  |     return self._send_loop(self.fd.send, data, flags)
shifthelper_webinterface_1  |   File "/opt/conda/lib/python3.5/site-packages/eventlet/greenio/base.py", line 371, in _send_loop
shifthelper_webinterface_1  |     return send_method(data, *args)
shifthelper_webinterface_1  | ConnectionResetError: [Errno 104] Connection reset by peer
shifthelper_webinterface_1  |
shifthelper_webinterface_1  | During handling of the above exception, another exception occurred:
shifthelper_webinterface_1  |
shifthelper_webinterface_1  | Traceback (most recent call last):
shifthelper_webinterface_1  |   File "/opt/conda/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1182, in _execute_context
shifthelper_webinterface_1  |     context)
shifthelper_webinterface_1  |   File "/opt/conda/lib/python3.5/site-packages/sqlalchemy/engine/default.py", line 470, in do_execute
shifthelper_webinterface_1  |     cursor.execute(statement, parameters)
shifthelper_webinterface_1  |   File "/opt/conda/lib/python3.5/site-packages/pymysql/cursors.py", line 166, in execute
shifthelper_webinterface_1  |     result = self._query(query)
shifthelper_webinterface_1  |   File "/opt/conda/lib/python3.5/site-packages/pymysql/cursors.py", line 322, in _query
shifthelper_webinterface_1  |     conn.query(q)
shifthelper_webinterface_1  |   File "/opt/conda/lib/python3.5/site-packages/pymysql/connections.py", line 834, in query
shifthelper_webinterface_1  |     self._execute_command(COMMAND.COM_QUERY, sql)
shifthelper_webinterface_1  |   File "/opt/conda/lib/python3.5/site-packages/pymysql/connections.py", line 1054, in _execute_command
shifthelper_webinterface_1  |     self._write_bytes(packet)
shifthelper_webinterface_1  |   File "/opt/conda/lib/python3.5/site-packages/pymysql/connections.py", line 1006, in _write_bytes
shifthelper_webinterface_1  |     raise err.OperationalError(2006, "MySQL server has gone away (%r)" % (e,))
shifthelper_webinterface_1  | pymysql.err.OperationalError: (2006, "MySQL server has gone away (ConnectionResetError(104, 'Connection reset b
y peer'))")
shifthelper_webinterface_1  |
shifthelper_webinterface_1  | The above exception was the direct cause of the following exception:
```
I do not know why this happens now, this is new.

MaxNoe, 2017-03-26:
Fixed now in https://github.com/fact-project/shifthelper_webinterface/pull/24 (currently deployed)

closed 2017-03-26 by MaxNoe