fact-project/shifthelper #229: shifthelper time is not UTC time anymore, off by 1 hour

relleums, 2017-03-26:
I got a call of shift helper at 02:40 UTC, but at least the shifthelper webpage shows the alarm call to be dropped at 03:40.

shift helper shows
2017-03-26 03:40:28  but according to smartfact and FACT++ it was 2017-03-26 02:40:28

MaxNoe, 2017-03-26:
Something to do with the daylight savings clock change?

dneise, 2017-03-27:
Sebastian you say:
> shift helper shows 2017-03-26 03:40:28 

What do you mean by this? The shifthelper website or the shifthelper logfile?
Because, I just checked the logfile and there is nicely shows the time in UTC:

> 2017-03-26 02:40:28,884 - shifthelper.tools.shift - DEBUG - Found shifter smueller
2017-03-26 02:40:29,149 - custos.notify.log - WARNING - Check: BiasVoltageOnButNotCalibrated - Message: There is a shift at the moment and 
Bias Voltage is On and 
 feedback is not calibrated 
2017-03-26 02:40:34,714 - shifthelper.tools.is_shift - DEBUG - is_shift_at_the_moment -> True

closed 2017-04-11 by MaxNoe