fact-project/shifthelper #230: Call during shutdown error?

madoee, 2017-04-07:
main.js crashed this morning during the shutdown due to DriveControl timeout. Shifthelper didn't call, I assume because it was past the scheduled shutdown time and it no longer checks for main.js to be running.
Since the shifter should be awake and monitoring the system, I don't know whether Shifthelper should even call, but it could help drawing the shifter's attention to the problem. So I'm reporting this to make sure this is expected behaviour and to discuss whether Shifthelper should call in this situation.

dneise, 2017-04-11:
I think this is indeed not the intended behaviour. 

Shifthelper goes into "daymode", i.e. it does not expect main.js to be running after the "Shutdown" task in the schedule. However ... the time of the "Shutdown" task is only the beginning of the actual shutdown ... not the end. So we expect main.js to be running, say another 10 minutes **after** the "Shutdown" task, right?

I guess this needs some work here:
https://github.com/fact-project/shifthelper/blob/master/shifthelper/tools/is_shift.py

MaxNoe, 2017-04-11:
I think this could be improved, but is no problem. The shifthelper makes sure the shifter is awake via the `I am awake` button. He has to supervise the shutdown. The shifthelper also makes sure the shutdown is completed via the checklist.

I don't know if we really want to call an awake shifter because something happens. Do we?

dneise, 2017-04-11:
I'm also not sure. I just would like to have the time range, within we want to somehow ensure that *main.js* is running to be defined. 

Of course main.js may run before the "Startup" task .. and it may also run after the shutdown has been completed ... but it actually *must* be running at least some time after the "Shutdown" task ... 

Still it is good that @madoee made us aware of this so we can clearly decide, what we want to do here.

MaxNoe, 2017-04-11:
It has to run some time after the shutdown task starts, yes. BUT: as the shutdown has to be supervised by an awake shifter, the shifthelper needs not to care in my opinion.

dneise, 2017-04-19:
So can we close this, or do we want to add some docu about this somewhere?