fact-project/shifthelper #236: Bad Gateway at log page.

mackaiver, 2017-05-17:
When I go to https://shifthelper.app.tu-dortmund.de/log I get the following error:

[image]

dneise, 2017-05-17:
Confirmed. Looks the same for me .... and I am nicely drunk ... :-D

MaxNoe, 2017-05-17:
MMh, this worked on the ihp-pc41 deployment, might have something to do with how this docker cloud works.

Will look into this.

dneise, 2017-09-05:
This was solved at some point, right? 
It was due to the local installation in dortmund right, and it did not need any changes in the shifthelper codebase right?

So Can I close this?

MaxNoe, 2017-09-05:
It was a wrong path in the config of the dortmund deployment

closed 2017-09-05 by MaxNoe