fact-project/shifthelper #238: FAD Loss results in call to Developer, not the shifter

MaxNoe, 2017-05-24:
Since a short time, a FAD loss results in a call to the developer because of IndexError exceptions as well as a call to the shifter because of Main.js not running.

Dimctrl Log:

```
 #> 01:11:47.209499 - Trying automatic reconnect [data taking,n=1]...
   ...disconnect 25
   ...waiting for 3s
   ...reconnect 25
   ...waiting for 1s
   ...checking connection
No crate to reset.
 E> 01:11:57.590678 - | scripts/takeRun.js: l.122: Error: Waiting for state "Idle" of server MCP [Connected] timed out. [main]
 E> 01:11:57.592604 - |     dim.wait("MCP", "Idle", 3000);
 E> 01:11:57.592886 - |         ^
 E> 01:11:57.599038 - |
 E> 01:11:57.599138 - | Error: Waiting for state "Idle" of server MCP [Connected] timed out.
 E> 01:11:57.599294 - |     at internal:1:245
 E> 01:11:57.599351 - |     at internal:1:336
 E> 01:11:57.599454 - |     at reconnect (scripts/takeRun.js:122:9)
 E> 01:11:57.599571 - |     at takeRun (scripts/takeRun.js:309:9)
 E> 01:11:57.599671 - |     at scripts/Main.js:1596:21
 E> 01:11:57.599787 - |     at main:1:1
 -> 01:11:57.649016 - State Transition from Running[3] to Idle[0] (0:scripts/Main.js[kbruegge:18377])
```

Shifthelper log:

```

```

dneise, 2017-09-05:
Is this still a problem?

MaxNoe, 2017-09-05:
I don't know. Wasn't developer in a long time now