fact-project/shifthelper #244: I was called even though I'm not on shift.

mackaiver, 2017-09-12:
Hello,

this morning I was called by the shifthelper. Both for the 'wakeup' call and the 'please fill shutdown checklist' call. Currently Andrii is on shift. (I don't know his github handle).

MaxNoe, 2017-09-12:
According to the shift calender, you **were** on shift

closed 2017-09-12 by MaxNoe

dneise, 2017-09-12:
I was just looking at the logs of the last days with

    cat shifthelper.log.2017-09-10 shifthelper.log | grep -v DEBUG | less

```
2017-09-10 20:43:07,164 - custos.notify.log - INFO - Check: DummyAlert - Message: Dummy Alert
2017-09-10 20:44:07,298 - custos.notify.log - INFO - Check: DummyAlert - Message: Dummy Alert
2017-09-11 05:33:33,015 - shifthelper.notifiers - INFO - Calling +49**707
2017-09-11 05:33:34,078 - custos.notify.log - WARNING - Check: IsUserAwakeBefore Shutdown - Message: There is a shift at the moment and 20min before shutdown and Parker not Awake
2017-09-12 05:30:44,272 - shifthelper.notifiers - INFO - Calling +49**707
2017-09-12 05:30:45,177 - custos.notify.log - WARNING - Check: IsUserAwakeBefore Shutdown - Message: There is a shift at the moment and 20min before shutdown and Parker not Awake
2017-09-12 05:32:44,394 - shifthelper.notifiers - INFO - Calling +49**707
2017-09-12 05:32:45,166 - custos.notify.log - WARNING - Check: IsUserAwakeBefore Shutdown - Message: There is a shift at the moment and 20min before shutdown and Parker not Awake
2017-09-12 06:00:25,048 - shifthelper.notifiers - INFO - Calling +49**707
2017-09-12 06:00:26,002 - custos.notify.log - WARNING - Check: ParkingChecklistFilled - Message: no shift at the moment and Last Shutdown is already 10min past and checklist not filled
```

dneise, 2017-09-12:
Looks like the Dummy Alert was not done on the evening of the 11th.