fact-project/shifthelper #246: Adjust condition `is_mainjs_not_running` to smartfact.cc bugfix

dneise, 2017-09-14:
Is last telcon a bug in smartfact.cc was found. Thomas proposed in the following email to fix it:

> Hallo,
> 
> also es scheint den Fall wirklich zu geben dass die Statuszeile leer ist... vermutlich weil mir der Fall nie untergekommen > ist und ich deswegen den status nicht in etwas "human readable" übersetzt habe.
> 
> Das passiert unter folgenden Bedingungen
> 
> Fall A:
> 
> 1) Rate Control nicht in SettingGlobalThreshold
> 2) Lid Control nicht in Moving
> 3) Rate Scan nicht in InProgress
> 4) MCP in (TriggerOn oder TakingData) und kein Configuration Name (das ist das was man beim MCP bei 'START' angibt) und > keine komplett gültige DRS Kalibration (bedeutet, dass von den drei möglichen runs (ped/gain/ped) der dritte noch nicht > genommen ist).
> 
> Fall B:
> 
> 1) identisch A1
> 2) identisch A2
> 3) identisch A3
> 4) MCP in einem der folgenden states:
>             kDimNetworkNA,
>             kDisconnected,
>             kConnecting,
>             kConnected,
>             kCrateReset0,
>             kCrateReset1,
>             kCrateReset2,
>             kCrateReset3,
> 
> Fall A wäre ein Bug, da (ausser man macht etwas manuell) immer ein Configuration Name gesendet wurde bevor das System in > TriggerOn oder TakingData ist.
> 
> 
> Zum Fall B:
> 
> Ich vermute die ersten vier States könnte man zusammenfassen zu soetwas wie "Not Ready" und die vier letzten "Resetting > crate"
> 
> Wenn da nichts dagegen spricht würde ich das so einbauen, ggf noch mit den Farben Rot (Not Ready) und Gelb "Resetting Crate".
> 
> Viele Grüße,
> Thomas.
> 

This would need an adjustment in [conditions.is_mainjs_not_running](https://github.com/fact-project/shifthelper/blob/master/shifthelper/conditions.py#L25)

I'd like to encourage fixing found bugs, even if this means adjustments down the line. 

I'll open a PR for this...

MaxNoe, 2017-09-14:
But it's not yet done, right? So we basically have this prepared and we merge it, when thomas fixes smartfact?

dneise, 2017-09-14:
it's not done yet. correct.